happycube/ld-decode #649: Vertical field shift - Causes line order to render incorrectly

IanSmallshire, 2021-07-15:
Sample file attached that causes the issue.

Symptom is that Most of the Sync on the Vsync are is maked as dropout.
[image]

Previous frame has a dropout on the last line but am unsure if this is the cause.
[image]

Zoomed in on the good frame the colourburst and VBI look correct as does the top left of the active image.
[image]

Zoomed in on the affected frame you can see the half video line is rendered second and the colourburst is wrong.
[image]

[AMvSync.zip](https://github.com/happycube/ld-decode/files/6824899/AMvSync.zip)

happycube, 2021-07-28:
Put a cut version of this into ld-decode-testdata: issues/649.ldf

happycube, 2021-07-28:
This appears to be triggered by a dropout in the hsync area of the last full line (617/619 from frame 6 on) which is confusing the sync detector.

IanSmallshire, 2021-07-28:
This seems to be a problem all across the disc. Do you require a larger sample to test against?

happycube, 2021-08-06:
Not yet... I'm still occasionally looking at it, I haven't cracked it yet.  Once I have a possible fix some more samples could be useful to see if it's a good one though.

happycube, 2021-09-11:
Getting field mismatch errors on another old sample (at fields 9-10, 23-24)

./ld-decode /mnt4/aiv/Volcanoes_DD86-DS1_Side1_20191227_CAV_PAL_00001-54000.ldf tmp -s 40000 -l 25 --pal

happycube, 2021-10-18:
> Currently decoding the full disc.... Will update when I have results

Oh, I forgot about the dropout... I don't know if I cracked that at all.
Sorry.  But it should get the fields right at least.

IanSmallshire, 2021-10-19:
Much improved and can also be seen in the output log files.
From my point of view can be pushed to master.

Here is the log from the start of the disc on current master branch.

`2021-10-18 14:26:23,783 - lddecode - DEBUG - Frame 1/110000: File Frame 109: CLV Timecode 0:xx 
2021-10-18 14:26:24,364 - lddecode - WARNING - At field #3, Field phaseID sequence mismatch (5->2) (player may be paused)
2021-10-18 14:26:24,365 - lddecode - DEBUG - Frame 2/110000: File Frame 110: CLV Timecode 0:xx 
2021-10-18 14:26:24,652 - lddecode - WARNING - At field #4, Field phaseID sequence mismatch (2->7) (player may be paused)
2021-10-18 14:26:24,935 - lddecode - WARNING - At field #5, Field phaseID sequence mismatch (7->4) (player may be paused)
2021-10-18 14:26:24,936 - lddecode - DEBUG - Frame 3/110000: File Frame 111: CLV Timecode 0:xx 
2021-10-18 14:26:25,222 - lddecode - WARNING - At field #6, Field phaseID sequence mismatch (4->1) (player may be paused)
2021-10-18 14:26:25,492 - lddecode - WARNING - At field #7, Field phaseID sequence mismatch (1->6) (player may be paused)
2021-10-18 14:26:25,493 - lddecode - DEBUG - Frame 4/110000: File Frame 112: CLV Timecode 0:xx 
2021-10-18 14:26:25,803 - lddecode - WARNING - At field #8, Field phaseID sequence mismatch (6->3) (player may be paused)
2021-10-18 14:26:26,084 - lddecode - WARNING - At field #9, Field phaseID sequence mismatch (3->8) (player may be paused)
2021-10-18 14:26:26,084 - lddecode - DEBUG - Frame 5/110000: File Frame 113: CLV Timecode 0:xx 
2021-10-18 14:26:26,323 - lddecode - WARNING - At field #10, Field phaseID sequence mismatch (8->5) (player may be paused)
2021-10-18 14:26:26,641 - lddecode - DEBUG - Frame 6/110000: File Frame 114: CLV Timecode 0:xx 
2021-10-18 14:26:27,187 - lddecode - WARNING - At field #13, Field phaseID sequence mismatch (7->4) (player may be paused)
2021-10-18 14:26:27,190 - lddecode - DEBUG - Frame 7/110000: File Frame 115: CLV Timecode 0:xx 
2021-10-18 14:26:27,495 - lddecode - WARNING - At field #14, Field phaseID sequence mismatch (4->1) (player may be paused)
2021-10-18 14:26:27,791 - lddecode - DEBUG - Frame 8/110000: File Frame 116: CLV Timecode 0:xx 
2021-10-18 14:26:28,379 - lddecode - DEBUG - Frame 9/110000: File Frame 117: CLV Timecode 0:xx 
2021-10-18 14:26:28,939 - lddecode - WARNING - At field #19, Field phaseID sequence mismatch (5->2) (player may be paused)
2021-10-18 14:26:28,940 - lddecode - DEBUG - Frame 10/110000: File Frame 118: CLV Timecode 0:xx 
2021-10-18 14:26:29,229 - lddecode - WARNING - At field #20, Field phaseID sequence mismatch (2->7) (player may be paused)
2021-10-18 14:26:29,487 - lddecode - DEBUG - Frame 11/110000: File Frame 119: CLV Timecode 0:xx 
2021-10-18 14:26:30,071 - lddecode - WARNING - At field #23, Field phaseID sequence mismatch (1->6) (player may be paused)
2021-10-18 14:26:30,071 - lddecode - DEBUG - Frame 12/110000: File Frame 120: CLV Timecode 0:xx 
2021-10-18 14:26:30,350 - lddecode - WARNING - At field #24, Field phaseID sequence mismatch (6->3) (player may be paused)
2021-10-18 14:26:30,606 - lddecode - WARNING - At field #25, Field phaseID sequence mismatch (3->8) (player may be paused)
2021-10-18 14:26:30,609 - lddecode - DEBUG - Frame 13/110000: File Frame 121: CLV Timecode 0:xx 
2021-10-18 14:26:30,936 - lddecode - WARNING - At field #26, Field phaseID sequence mismatch (8->5) (player may be paused)
2021-10-18 14:26:31,184 - lddecode - WARNING - At field #27, Field phaseID sequence mismatch (5->2) (player may be paused)
2021-10-18 14:26:31,186 - lddecode - DEBUG - Frame 14/110000: File Frame 122: CLV Timecode 0:xx 
2021-10-18 14:26:31,476 - lddecode - WARNING - At field #28, Field phaseID sequence mismatch (2->7) (player may be paused)
2021-10-18 14:26:31,717 - lddecode - WARNING - At field #29, Field phaseID sequence mismatch (7->4) (player may be paused)
2021-10-18 14:26:31,717 - lddecode - DEBUG - Frame 15/110000: File Frame 123: CLV Timecode 0:xx 
2021-10-18 14:26:32,039 - lddecode - WARNING - At field #30, Field phaseID sequence mismatch (4->1) (player may be paused)
2021-10-18 14:26:32,321 - lddecode - WARNING - At field #31, Field phaseID sequence mismatch (1->6) (player may be paused)
2021-10-18 14:26:32,322 - lddecode - DEBUG - Frame 16/110000: File Frame 124: CLV Timecode 0:xx 
2021-10-18 14:26:32,617 - lddecode - WARNING - At field #32, Field phaseID sequence mismatch (6->3) (player may be paused)
2021-10-18 14:26:32,841 - lddecode - DEBUG - Frame 17/110000: File Frame 125: CLV Timecode 0:xx 
2021-10-18 14:26:33,466 - lddecode - DEBUG - Frame 18/110000: File Frame 126: CLV Timecode 0:xx 
2021-10-18 14:26:34,020 - lddecode - DEBUG - Frame 19/110000: File Frame 127: CLV Timecode 0:xx 
2021-10-18 14:26:34,601 - lddecode - DEBUG - Frame 20/110000: File Frame 128: CLV Timecode 0:xx  `

Here is the start of the log on the 649 test branch

`2021-10-18 16:25:29,363 - lddecode - DEBUG - Frame 1/110000: File Frame 109: CLV Timecode 0:xx 
2021-10-18 16:25:29,984 - lddecode - DEBUG - Frame 2/110000: File Frame 110: CLV Timecode 0:xx 
2021-10-18 16:25:30,580 - lddecode - DEBUG - Frame 3/110000: File Frame 111: CLV Timecode 0:xx 
2021-10-18 16:25:31,163 - lddecode - DEBUG - Frame 4/110000: File Frame 112: CLV Timecode 0:xx 
2021-10-18 16:25:31,710 - lddecode - DEBUG - Frame 5/110000: File Frame 113: CLV Timecode 0:xx 
2021-10-18 16:25:32,274 - lddecode - DEBUG - Frame 6/110000: File Frame 114: CLV Timecode 0:xx 
[... cut ...]
2021-10-18 16:25:37,759 - lddecode - DEBUG - Frame 15/110000: File Frame 123: CLV Timecode 0:xx 
2021-10-18 16:25:38,439 - lddecode - DEBUG - Frame 16/110000: File Frame 124: CLV Timecode 0:xx 
2021-10-18 16:25:39,028 - lddecode - DEBUG - Frame 17/110000: File Frame 125: CLV Timecode 0:xx 
2021-10-18 16:25:39,625 - lddecode - DEBUG - Frame 18/110000: File Frame 126: CLV Timecode 0:xx 
2021-10-18 16:25:40,230 - lddecode - DEBUG - Frame 19/110000: File Frame 127: CLV Timecode 0:xx 
2021-10-18 16:25:40,801 - lddecode - DEBUG - Frame 20/110000: File Frame 128: CLV Timecode 0:xx `

happycube, 2022-01-02:
Can't close this yet since the vblank dropouts are still there.

I may reassign this to rev8 if I can't figure it out in time though.

happycube, 2022-01-30:
Found an NTSC sample where an entire vsync area dropped out as well, added to testdata

closed 2022-02-02 by happycube