happycube/ld-decode #651: PAL- Amplitude of Left and Right Analogue audio is different

IanSmallshire, 2021-07-16:
This is from a disc labelled as Mono.

Raw PCM 
[image]

Causes the CX decoded audio to clip
[image]

happycube, 2021-08-25:
[image]

I don't see a difference in PAL GGV, but I should cut amplitude by 3db.

ValZapod, 2021-09-22:
In music it is possible to have even sound in only one channel or changing. This looks like music(?)

happycube, 2021-10-17:
I did cut the amplitude, and since this appears to be a mono disk and the two channels are in pretty good sync on PAL GGV, I consider this PDO weirdness.

closed 2021-10-17 by happycube