happycube/ld-decode #670: Review use of JsonWax in ld-decode tools

simoninns, 2021-12-05:
JsonWax library git looks static with no updates since Dec 2018:

https://github.com/doublejim/JsonWax

This will soon start to create compilation issues with newer Ubuntu and Qt versions. Options are fork and modify or find another library.

atsampson, 2021-12-05:
JsonWax is also **incredibly** inefficient, so I'd be very much in favour of replacing it with a proper parser.

A couple of years ago I experimented with using Qt's own JSON parser instead - [unfinished patch here](https://github.com/atsampson/ld-decode/commit/d516b8baa3927dd680800a4e985076f46842227b) (I think I stopped because I wasn't sure whether it should preserve data in the JSON file that it didn't recognise - which makes it a bit more complicated).

simoninns, 2021-12-05:
I'm going to have to go back and review the original code but, from memory, I designed it so all the tools pull the JSON parsing functions from a central library function.  Therefore it shouldn't matter about preserving what's there as all the code is in sync (except for the python ld-decode, but that's the case as-is).  I'll review it all and make some decisions.

simoninns, 2021-12-05:
As a side note - the original Qt JSON routines would crash once the file size got above a certain limit (way below what was needed for ld-decode) - that was the reason why JsonWax was used instead.  Hopefully those restrictions have been fixed now.

oyvindln, 2021-12-05:
Jsonwax uses the Qt functions internally I think so presumably it should work?

simoninns, 2021-12-11:
@atsampson - I've managed to get the tool-chain reading JSON using the in-built library, but I'm unable to figure out how I 'find' a particular field and modify it's data (without pulling the whole field array into memory, deleting the original and re-inserting it).

Do you have any practical knowledge of modifying nested JSON data without re-writing/re-ordering everything... I can't seem to find any useful examples :/

simoninns, 2021-12-11:
This doesn't fill me with confidence: https://bugreports.qt.io/browse/QTBUG-25723?focusedCommentId=184919

simoninns, 2021-12-12:
ok, my conclusion is that the in-built JSON library is still not capable enough to deal with the requirements in the tool chain; specifically, it has no meaning full API to allow modification of nested parameters without pulling the whole fields array into memory (which, like in the past, would crash the library).

Recommendation is remain as-is or open a new issue request to port over to RapidJSON as a more well supported 3rd party solution.

closed 2021-12-12 by simoninns