happycube/ld-decode #671: Installation of ld-decode under Ubuntu 21.10

simoninns, 2021-12-05:
Similar to issue #668 (for ld-decode-tools) there seem to be issues running ld-decode under Ubuntu 21.10.

Currently Ubuntu 20.04 LTS is the supported OS - however, Ubuntu 22.04 LTS is now 4 months away... probably a good time to start looking into what's needed to maintain support for the next LTS.

simoninns, 2021-12-05:
Running ld-decode with no command parameters results in the following output on 21.10:

```
sdi@sehoapollo01:~/github/ld-decode$ ld-decode
Traceback (most recent call last):
  File "/usr/local/bin/ld-decode", line 4, in <module>
    __import__('pkg_resources').run_script('ld-decode==7', 'ld-decode')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1455, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.9/dist-packages/ld_decode-7-py3.9.egg/EGG-INFO/scripts/ld-decode", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 627, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.9/dist-packages/ld_decode-7-py3.9.egg/lddecode/audio.py", line 25, in <module>
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 627, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.9/dist-packages/ld_decode-7-py3.9.egg/lddecode/core.py", line 32, in <module>
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 627, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.9/dist-packages/ld_decode-7-py3.9.egg/lddecode/utils.py", line 24, in <module>
  File "/usr/lib/python3/dist-packages/numba/core/decorators.py", line 214, in wrapper
    disp.enable_caching()
  File "/usr/lib/python3/dist-packages/numba/core/dispatcher.py", line 781, in enable_caching
    self._cache = FunctionCache(self.py_func)
  File "/usr/lib/python3/dist-packages/numba/core/caching.py", line 616, in __init__
    self._impl = self._impl_class(py_func)
  File "/usr/lib/python3/dist-packages/numba/core/caching.py", line 351, in __init__
    raise RuntimeError("cannot cache function %r: no locator available "
RuntimeError: cannot cache function 'scale': no locator available for file '/usr/local/lib/python3.9/dist-packages/ld_decode-7-py3.9.egg/lddecode/utils.py'

```

oyvindln, 2021-12-05:
Hm, I'm running it on Ubuntu with no issue. Haven't used the install though, just run it from a local directory.

simoninns, 2021-12-05:
I did this from a not-yet-paint-dry fresh install... and the error I quoted was the result :)  So, I'm quite sure there is an issue.  Looks like something to do with the numpty library; but python error messages are not exactly human readable!  My plan is to set up a virtualbox install and then snapshot it before installing to make testing easy.

For installation, I followed the wiki and didn't try to be clever.  The Qt based tools didn't install and compile either - so there is another issue for that, plus the job of updating the wiki

simoninns, 2021-12-05:
I just tried a fresh 20.04.03 LTS installation and ran through the install wiki again.  The 'make' package was missing (added to the wiki already), but I got a similar python error from ld-decode as with my 21.10 test... so I don't think this python error is local to the OS version.  Here's the output from a brand-new install on 20.04:

```
sdi@sehoapollo01:~/github/ld-decode$ ld-decode
Traceback (most recent call last):
  File "/usr/local/bin/ld-decode", line 4, in <module>
    __import__('pkg_resources').run_script('ld-decode==7', 'ld-decode')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1470, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/ld_decode-7-py3.8.egg/EGG-INFO/scripts/ld-decode", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/dist-packages/ld_decode-7-py3.8.egg/lddecode/audio.py", line 25, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/dist-packages/ld_decode-7-py3.8.egg/lddecode/core.py", line 32, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/dist-packages/ld_decode-7-py3.8.egg/lddecode/utils.py", line 24, in <module>
  File "/usr/lib/python3/dist-packages/numba/decorators.py", line 193, in wrapper
    disp.enable_caching()
  File "/usr/lib/python3/dist-packages/numba/dispatcher.py", line 679, in enable_caching
    self._cache = FunctionCache(self.py_func)
  File "/usr/lib/python3/dist-packages/numba/caching.py", line 614, in __init__
    self._impl = self._impl_class(py_func)
  File "/usr/lib/python3/dist-packages/numba/caching.py", line 348, in __init__
    raise RuntimeError("cannot cache function %r: no locator available "
RuntimeError: cannot cache function 'scale': no locator available for file '/usr/local/lib/python3.8/dist-packages/ld_decode-7-py3.8.egg/lddecode/utils.py'

```

atsampson, 2021-12-05:
I think this is fallout from the change I made in #667 to use setuptools rather than distutils. I'll do a PR to revert the change.

closed 2021-12-05 by happycube

simoninns, 2021-12-05:
Yup; that seems to have cured it :)  Thanks

oyvindln, 2022-07-16:
Are you able to reproduce this issue still? 
Was testing in a 20.04 VM and couldn't get this to happen when swapping to setuptools in setup.py. (Works fine on my main computer with 22.04 too for that matter, but it's probably been fixed in the deps there).
Seems it's some issue with the cache folder not being writeable and has some workarounds:
https://github.com/numba/numba/issues/4032

We need to move to setuptools eventually, and I've been meaning to look into publishing ld-decode/vhs-decode on pip for easy installation so would be helpful if this could be sorted.

PR #672 (2021-12-05): Revert "Use setuptools rather than distutils."
This reverts commit 80660adbad19b4ba739d3ff71e720f28766d1bf1 (#667).

Installation using setuptools doesn't work properly on Ubuntu 21.10 with Python 3.8, so we can't switch to this yet.

Fixes #671 (hopefully).