happycube/ld-decode #677: isWidescreen metadata field in videoParameters

simoninns, 2021-12-11:
From wiki: isWidescreen | Boolean | true if the video is 16:9 anamorphic, rather than 4:3

How does ld-decode know what to set here... I don't see how this metadata field would ever get set to 'true'.

Unless the intention is made clear, I vote to remove it.

atsampson, 2021-12-11:
This could be set automatically if there's [WSS](https://en.wikipedia.org/wiki/Widescreen_signaling) in the video, e.g. by ld-process-vbi.

But I'd also like to be able to override some of the video parameters in ld-analyse - particularly for adjusting black/white level - in which case this would be another useful parameter to be able to set by hand for later processing.

simoninns, 2021-12-11:
Generally I would steer away from introducing flags until they are useful for something... however, there do seem to be a whole bunch of new metadata fields which are simply hopeful that someone will implement something with them someday; so it's not a big deal really.

Alternatively we could remove the flag and raise an issue for WSS detection support in ld-process-vbi so both are implemented at the same time.

Let me know which you prefer and I'll close the issue.

atsampson, 2021-12-11:
isWidescreen was introduced because it is already useful for something - ld-chroma-decoder needs to know the aspect ratio when generating a Y4M header. The future work here is just having a more convenient way of setting it rather than needing to edit the JSON by hand. So I'd much rather keep it.

oyvindln, 2021-12-11:
ld-analyse also has an option for showing in anamorphic widescreen, though I'm not sure if it auto-sets it from the flag yet.

simoninns, 2021-12-11:
ok, np for me

closed 2021-12-11 by simoninns