happycube/ld-decode #680: frameNumber parameter in 'field' metadata JSON

simoninns, 2021-12-12:
There seems to be a frameNumber parameter in the JSON field object.

It's not in the wiki.

Looks like a bug as fields don't technically have a frame number...

It can't be the field number from the input file, because that's "diskLoc"...

`
{"fields":[{"audioSamples":738,"decodeFaults":0,"diskLoc":583.4,"efmTValues":13641,"fieldPhaseID":1,"fileLoc":389350760,"isFirstField":true,"medianBurstIRE":18.931,"ntsc":{"ccData0":0,"ccData1":0,"fieldFlag":true,"fmCodeData":327680,"isFmCodeDataValid":true,"whiteFlag":true},"seqNo":1,"syncConf":100,"vbi":{"vbiData":[9101312,15785216,15785216]},"vitsMetrics":{"bPSNR":39.8,"wSNR":37.3}},{"audioSamples":734,"decodeFaults":0,"diskLoc":584.4,"efmTValues":14670,"fieldPhaseID":2,"fileLoc":390019502,"frameNumber":0,"isFirstField":false,"medianBurstIRE":18.817,"ntsc":{"ccData0":-1,"ccData1":-1,"fieldFlag":false,"fmCodeData":851968,"isFmCodeDataValid":true,"whiteFlag":false},"seqNo":2,"syncConf":100,"vbi":{"vbiData":[9289728,8912895,8920541]},"vitsMetrics":{"bPSNR":40}},{"audioSamples":737,"decodeFaults":0,"diskLoc":585.4,"efmTValues":14730,"fieldPhaseID":3,"fileLoc":390685511,"isFirstField":true,"medianBurstIRE":19.24,"ntsc":{"ccData0":0,"ccData1":0,"fieldFlag":true,"fmCodeData":851968,"isFmCodeDataValid":true,"whiteFlag":true},"seqNo":3,"syncConf":100,"vbi":{"vbiData":[9101313,15785216,15785216]},"vitsMetrics":{"bPSNR":39.8,"wSNR":34.7}},{"audioSamples":735,"decodeFaults":0,"diskLoc":586.4,"efmTValues":14627,"fieldPhaseID":4,"fileLoc":391354209,"frameNumber":1,"isFirstField":false,"medianBurstIRE":19.017,"ntsc":{"ccData0":-1,"ccData1":-1,"fieldFlag":false,"fmCodeData":851968,"isFmCodeDataValid":true,"whiteFlag":false},"seqNo":4,"syncConf":100,"vbi":{"vbiData":[9289728,8912895,8920541]},"vitsMetrics":{"bPSNR":39.7}},{"audioSamples":737,"decodeFaults":0,"diskLoc":587.4,"efmTValues":14701,"fieldPhaseID":1,"fileLoc":392020199,"isFirstField":true,"medianBurstIRE":19.049,"ntsc":{"ccData0":0,"ccData1":0,"fieldFlag":true,"fmCodeData":851968,"isFmCodeDataValid":true,"whiteFlag":true},"seqNo":5,"syncConf":100,"vbi":{"vbiData":[9101314,15785216,15785216]},"vitsMetrics":{"bPSNR":40,"wSNR":35.6}}
`

happycube, 2021-12-26:
It's written if it's a CAV field with a frame # in the VITS.  It can be used to determine if it's the start of a pulldown frame on an NTSC disk.  Added to the wiki.

simoninns, 2021-12-27:
This doesn't seem like a good idea... the field order is not readily determined so it will be wrong if reversed.  The tools-library determines the frame number based on the VBI and reverse field command line option and this is 'on the fly' since stacking and other processing can improve the information accuracy.

Where in the VITS is this information (so I can check the specs)?

happycube, 2021-12-27:
Section 10.1.3 of IEC 68057 (non-amendment) states that the VITS frame # is inserted into whichever field is the first of the CLV picture.  

I could stuff this behind the verbose-vits where the CLV equivalent data is.

simoninns, 2021-12-27:
If it's in the VITS spec then I'm ok with it, but the field name is going to confuse the casual viewer... either pop it in verbose-vits or change it to vitsFrameNumber to make it clearer.  The majority of the very optional data (from a mastering point of view) isn't used by the tools; so additional metadata is just increasing JSON loading times so I would go the verbose-vits route (since any field data has a big impact on the JSON file-size).

happycube, 2022-01-18:
It's behind verboe-vits now.

closed 2022-01-18 by happycube