happycube/ld-decode #707: Chroma Decoding broken in ld-analyuse

IanSmallshire, 2022-02-28:
Using commit f40b68302cde42d03d737486776d9c18c11ef186 ( Feb 7, 2022) the colour is available in ld-analyse
Using the latest commit 7e6eac590c06b01ed1ca9aac88dd3b2bf734677c (Feb 7, 2022) the image is black when viewing chroma.

Old Commit
[image]
[image]

New version (showing chroma)
[image]

IanSmallshire, 2022-02-28:
Have checked it was broken on this commit 08f32a5782117dd5f3125b80219f7f01bf2f527e

IanSmallshire, 2022-02-28:
Nothing showing up with starting ld-analyse -d for the debug options..
[image]

atsampson, 2022-02-28:
@MooglyGuy, would you mind having a look at this? Probably missing initialisation in ld-analyse for your padding change - if you can't figure out what's wrong let me know and I'll have a look later.

MooglyGuy, 2022-03-01:
Absolutely, I'll have a look at it this evening. I do apologize for the delay, work and life have been chaos.

atsampson, 2022-03-01:
Don't worry, I know the feeling!

IanSmallshire, 2022-03-01:
Confirmed fix, Cheers

commit cda86e87 (2022-03-18): fix #707/709 in the library itself
closed 2022-03-01 by IanSmallshire