happycube/ld-decode #716: ld-chroma-decoder not finding json file

titan-91, 2022-04-27:
When trying to run ld-chroma-decoder with the required tbc and json files present after a decode, I am getting the below terminal output:

Critical: JSON wax library error: "No errors occured."
Critical: Opening JSON file failed: JSON file cannot be opened/does not exist
Info: Unable to open ld-decode metadata file

Has there been a change or regression with JsonWax?

atsampson, 2022-04-27:
I've just built master, and the test suite passes happily, so I don't think anything's broken.

The error you're getting is what you see if it can't open the input files at all - are you sure it's being invoked with the right filename (e.g. it's not missing the `.tbc` or similar)?

titan-91, 2022-04-28:
Yes, the command I'm using seems to be correct, unless something changed with the syntax. Screenshot attached.

[image]

closed 2022-04-28 by titan-91