happycube/ld-decode #726: Remove videoParameters.fsc from JSON

atsampson, 2022-05-18:
Currently the JSON `videoParameters` object contains a field called `fsc`: "The fSC in Hz, truncated to an integer".

The "truncated" part is important, because for both NTSC and PAL the true value has a fractional part. The few places in the tools where it's used expect it to be the correct value, so they're very slightly wrong at the moment - and for PAL this does actually make a small difference to the output.

Since the value of fSC is fixed for each video standard we support, I'm going to suggest that we:
- stop ld-decode from writing `fsc` at all, and remove it from the documentation
- make `VideoParameters::fsc` a double, rather than an integer, and rename it to `fSC`
- make the JSON parser ignore the fsc field if present, and set `fSC` exactly based on the video standard (as it does for `LineParameters` and other per-standard information)

I'd also like to make the audio and video `sampleRate` parameters doubles at the same time - in all but one place in the code they're used in a double context, and there's no reason why you couldn't have a non-integer sample rate.

The tools will still handle existing JSON files correctly after this change.

Any thoughts?

simoninns, 2022-05-19:
I don't see any problem with this - looks logical. IIRC double wasn't supported by JSON which is why it wasn't used.  Has this changed?

We still have issue #242 to consider also... Not sure if that has meaning here since, currently, I don't think ld-decode is following the video standard (in a very technical sense)?

simoninns, 2022-05-19:
Great; then it's an 'ok' from me - I think it's a good idea

oyvindln, 2022-05-19:
Unless we need a custom fsc for some reason one could as well chose it from a constant based on the video standard and line count/sample rate, so it could maybe even be left out of the JSON entirely unless needed. (or optionally supplied as some string id in those cases.)

atsampson, 2022-05-21:
See discussion in #727 - I'm now not sure that removing fSC makes sense until we've got a better story about how to support more than the existing two video standards.

closed 2022-05-21 by happycube

happycube, 2022-10-11:
Sounds good to me too.

PR #727 (2022-05-21): Use double values for sampleRates and fSC
Fixes #726. This is built on a slightly-refactored version of MooglyGuy's LineParameters work - the big changes are in the last commit:

Use double values for `sampleRate` and `fSC`. Several of the tools want exact values for these.

For video `sampleRate`, previously ld-decode output `int(fSC) * 4` to the JSON, while actually using `fSC * 4` internally. Output the full floating-point value of `fSC * 4` instead. Correct ld-chroma-encoder to match this.

The audio `sampleRate` has always been an integer anyway in ld-decode.

For `fsc`, previously ld-dec [... cut]