happycube/ld-decode #731: Compiler warnings

simoninns, 2022-05-28:
Seems like there are some compiler warning - I guess it is caused by some recent changes.  Just making a note here:

`g++ -c -pipe -O2 -std=gnu++11 -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DAPP_BRANCH="\"master\"" -DAPP_COMMIT="\"35ff6aa\"" -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I. -I../library/filter -I../library/tbc -I../ld-chroma-decoder -I/usr/include/qwt -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o tbcsource.o tbcsource.cpp
In member function ‘void QVector<T>::append(const T&) [with T = Frame]’,
    inlined from ‘void DiscMap::addPadding(qint32, qint32)’ at discmap.cpp:642:24:
cc1plus: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qbytearray.h:46,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:50,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qcoreapplication.h:44,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QCoreApplication:1,
                 from discmap.h:28,
                 from discmap.cpp:25:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qarraydata.h: In member function ‘void DiscMap::addPadding(qint32, qint32)’:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qarraydata.h:129:8: note: at offset 24 into destination object ‘QTypedArrayData<Frame>::<anonymous>’ of size 24
  129 | struct QTypedArrayData
      |        ^~~~~~~~~~~~~~~
In member function ‘void QVector<T>::append(const T&) [with T = Frame]’,
    inlined from ‘void DiscMap::addPadding(qint32, qint32)’ at discmap.cpp:642:24:
cc1plus: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
/usr/include/x86_64-linux-gnu/qt5/QtCore/qarraydata.h: In member function ‘void DiscMap::addPadding(qint32, qint32)’:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qarraydata.h:129:8: note: at offset [25, 49] into destination object ‘QTypedArrayData<Frame>::<anonymous>’ of size 24
  129 | struct QTypedArrayData
      |        ^~~~~~~~~~~~~~~
`

Generated after a git pull

`From ssh://github.com/happycube/ld-decode
   f9b6be1..35ff6aa  master     -> origin/master
Updating f9b6be1..35ff6aa
`

oyvindln, 2022-05-28:
Would have to test, but it's possible it has something to do with the constructor of the Frame class having const reference arguments with default parameters so you end up with some code that invokes references to "magic numbers" which might not be entirely ideal. The constructor params don't really need to be by reference references anyhow since the values are all int/float/bools.

simoninns, 2022-05-29:
Running the latest Qt creator, I can recreate the compiler output by switching into release mode.  The warning seems to be due to [-Wstringop-overflow=] and possibly the -O2 (which I think the string overflow is introducing.

I am struggling to understand the meaning of the error though which is caused when the discmapper appends a padding frame to a QVector of Frame objects (line 642 of discmap.cpp)

simoninns, 2022-05-29:
I'm pretty sure this is related to:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101854

There are no string operations taking place, so the compiler warning really makes no sense.

simoninns, 2022-05-29:
I'm going to won't-fix this - if the warning remain then I will re-open the issue

closed 2022-05-29 by simoninns