happycube/ld-decode #758: ld-chroma-decoder lacks comprehensive documentation

simoninns, 2022-06-12:
The ld-chroma-decoder has grown exponentially in complexity and the number of both LD and VHS decode use-cases it can provide.

It's time for it to have it's own manual with clear explanations of the options (with examples if possible) as well as a general how-you-do-it guide for users less familiar with the chroma decoding processes.

simoninns, 2022-06-13:
I've now split all the tools into individual pages; so the chroma decoder documentation has a place to live (see the ld-analyse page for an example)

atsampson, 2022-08-19:
The documentation is pretty detailed now - the options are all described on the ld-analyse page (because that's usually the best way to experiment with them), and there's a link to that section on the ld-chroma-decoder page.

atsampson, 2022-08-20:
Updated the chroma decoder bits in the manual following the recent changes - I think this is done now.

closed 2022-08-20 by atsampson