happycube/ld-decode #769: Support for more Linux distros

lizardgai4, 2022-08-09:
This will be a living document, updated as I try installing it on various Linux distros.

**Debian family:**

Since it was designed for Ubuntu, this is a low priority issue.  There are people who got it to work on Ubuntu and Mint.  I will attempt to install it on Pop!_os sometime.

**Red Hat Family:**

**Fedora:** I tried installing dependencies from DNF only.  samplefmt.h is located in usr/include/ffmpeg/libavutil/samplefmt.h, but make is expecting it in the Ubuntu location /usr/include/x86_64-linux-gnu/libavutil/samplefmt.h

Although Fedora can use apt, it's not natively installed, and people probably installed Fedora for dnf usage.

**Arch Family:**

**Manjaro:** To be tried

**OpenSUSE:**

To be tried

DreckSoft, 2022-08-14:
ld-decode actually works on my Manjaro. Just needed to change LIBS += -lqwt-qt5 to LIBS += -lqwt in ld-analyse.pro which is already mentioned in a commentary above that line.

Unfortunately I can't remember which packaged I did install.

lizardgai4, 2022-08-14:
@DreckSoft
"Unfortunately I can't remember which packaged I did install."

That's okay.  I figure out which packages to install on each Linux distro, using the preferred package managers (Ubuntu has apt, Fedora has dnf, Manjaro has pacman, etc.)

lizardgai4, 2022-08-24:
I am currently working on CMake files for ld-decode, and got ld-chroma-decoder to compile and give display the help screen.

lizardgai4, 2022-09-02:
There is currently a pull request to have this project use CMake files, which work on nearly every Linux distro I tried (see the document link at the top for details).  Can we mark this as resolved just yet?

atsampson, 2022-09-02:
If you put "Fixes #769." in the message for the PR, then this issue will be closed automatically when the PR is merged.

lizardgai4, 2022-09-02:
@atsampson Got it, gave it that message

PR #780 (2022-09-04): First CMake files
Fixes #769 

Plus new gitignore parameters for CMake-generated files,  It's a good start.

Current flaws:
- The program only works on Linux (but porting is very hard)

Fixed flaws:
- Requires code modifications to work on multiple Linux distros
- ~~-ld-analyse requires modifications to the generated Makefile to compile~~ Fixed (on Ubuntu at least)
- FindQWT code and ld-analyse code currently occupy the same CMakeLists.txt file.
- Having Qt6 installed at all might break installation (explicitly mentioning Qt5 instead of just Qt in the CMake files may fix it)
- library install  [... cut]
closed 2022-09-04 by happycube