happycube/ld-decode #781: ld-disc-stacker crash issue

vrunk11, 2022-08-26:
So I have 3 discs but 1 of them have a big scratch that keep the player skipping back so I made 2 sample for capturing the whole disc

when I try to stack the disc, it read the smaller sample (the part 1 on the scratched disc) with a wrong length
and when it reaches the end of the sample it crash

here is the link with the 3 tbc files where I can reproduce the crash (using a Windows build) :
[https://drive.google.com/file/d/1NK10nCDHgFBhN6rso7621Uo2N7TNwQcr/view?usp=sharing](url)

simoninns, 2022-08-27:
A couple of things - you have linked TBC files - recreation of the error (since there is no indication of where in the tool chain the JSON gets corrupted) requires end-to-end.  Therefore you need to supply laserdisc samples (as the JSON is created by ld-decode, so any bugs there will already be in the TBC files).

Also it would be useful if you could include the steps to recreate.  From the LDS until the error occurs - what commands did you use (on the linked LDS files) to make the problem appear.

Also state any environmental things (which OS version and build, etc.)

vrunk11, 2022-08-28:
here is an ldf cuted with the part that cause issue [https://drive.google.com/file/d/1uOlufL4f8HilODcnzgCvS8U_8ab8qEnP/view?usp=sharing](url)
i decoded using NTSC option ld-decode then  -> process-vbi ->disc-map -> disc-stacker 
(on Windows 10)

simoninns, 2022-08-28:
FIxed in latest PR to master

closed 2022-08-28 by simoninns