happycube/ld-decode #782: Add OS information into logs/debug output for ld-decode-tools and ld-decode

simoninns, 2022-08-27:
Since there are many users running in unsupported environments it is getting really hard to tell if issue reports are really issues at all.  The ld-decode tools (and probably ld-decode) should report the underlying OS and version in all debug so if bugs are reported on unsupported systems they can be pushed to the port maintainer for first-pass analysis.

happycube, 2022-08-29:
Done.

simoninns, 2022-09-03:
I need to modify the tools output too... So note when ld-decode is done and I'll update when ld-decode-tool match... then we can close.

harrypm, 2023-12-07:
It would be good if this also stores inital command used for decoding aswell so user error can be easily catched/pointed out.

simoninns, 2025-12-05:
Closing this - the tools and ld-decode have moved on - so this issue is not current.   Make a new issue referring to the 2025 code if this is still seen as an issue