happycube/ld-decode #783: Misaligned fields on Clerks side 2

atsampson, 2022-08-27:
(This is probably an after-rev7 kind of problem.)

Using current master, with a Technidisc-pressed copy of [Clerks (3618 AS)](https://www.lddb.com/laserdisc/06383/3618-AS/Clerks:-Directors-Edition), side 1 decodes happily, but side 2 - which is quite long, at 59m49s - produces errors throughout:

```
Frame 1/110000: File Frame 1: CLV Timecode CLV Timecode 0:00.00 frame 0         
At field #2, Field phaseID sequence mismatch (2->1) (player may be paused)
At field #3, Field phaseID sequence mismatch (1->4) (player may be paused)
Frame 2/110000: File Frame 2: CLV Timecode CLV Timecode 0:00.01 frame 1         
At field #4, Field phaseID sequence mismatch (4->3) (player may be paused)
At field #5, Field phaseID sequence mismatch (3->2) (player may be paused)
[...]
```

Looking at the TBC, this is because some fields are being offset to the right by 2 samples. The video signal appears to have fairly ringy transitions on the edges of the sync pulses, so I wonder if the TBC is locking to an incorrect edge...

[image]

I've put samples of both sides on the SFTP under `/sftp/adamsampson/Faults/issue783`.

Gamnn, 2022-08-28:
Similar issue on this sample, worked around with --video_lpf 4.7
@happycube asked me to note it here.

`/sftp/chadpage/issues/i783-phase.ldf`

atsampson, 2022-08-29:
It's all OK if I decode with `--video_lpf 4.0` -- if I go much higher than that then I get increasing numbers of misaligned fields.

happycube, 2022-08-29:
4.0 is very low - I was able to use 4.7 on the sample you posted, can you post a more difficult one?

atsampson, 2022-08-29:
I've put up a couple more clips - `3618as-side2-middle1.ldf` needs 4.4, `3618as-side2-start.ldf` needs 4.2, and `3618as-side2-middle2.ldf` needs 4.1.

simoninns, 2025-12-09:
Is this issue still an issue?  No update since 2022?