happycube/ld-decode #826: Add WSS (Wide Screen Signaling) Support for ld-process-vbi

harrypm, 2023-01-03:
Was just reading over a couple of things and noticed issue #677 which support was not implemented for WSS, so I am making this issue to add WSS to `ld-process-vbi` 

This would be incredibly useful on the tape side not only to set flags properly in gen_chroma_vid.sh scripts, but also as some tapes are recorded in the cropped letterbox format.

Most fully equipped colour bar generators can flag 16:9/4:3 like ones found in HDV cameras with Composite/S-Video out for legacy use.

## WSS Basic Flags

* full format 4:3

* full format 16:9 (anamorphic)

* letterbox

* pillar-box

* zoomed letterbox

## WSS Full Flags 

[image]

## Raster Example 

[image]

## Technical 

Line 23 is its standard position typically observed.

[EN 300 294 V1.3.2 (1998-04)](https://drive.google.com/file/d/1NxInDRUjqwMWRO-bPhlvjoDdAMTcWkCK/view?usp=sharing)

[Rec. ITU-R BT.1119-2 (1994-1995-1998)](https://drive.google.com/file/d/1rNbw4W7QkFQqij7EgMFXB2w5mJzJ_n5f/view?usp=share_link)

ifb, 2023-01-15:
What do you mean by "Add writing support to `ld-chroma-encoder`"?  The only way it has to signal any sort of metadata (PAR or otherwise) is when you use y4m.  That code is already there.

harrypm, 2023-01-15:
@ifb I mean literally generating based on the file or by manual flags, then embedding WSS lines into line 23 or higher. 

It's a small thing but a relatively large upgrade for test media generation, and correct metadata flagging.

ifb, 2023-01-16:
Oh, I read that (several times) as ld-chroma-**de**coder.

WSS reading isn't too far different from closed captions.  @atsampson might be interested in adding it.  XDS (in closed captions) can signal aspect ratio too, but I don't think that was ever widely used.  V-chip and time of day (for automatically setting VCR time) was its main use case.

Writing WSS is pretty obscure.  Maybe someone will work on it.  Personally, I'd want to add an SCC writer but that'd be more involved.

harrypm, 2023-01-16:
Well, it's a simple repeating pattern to replicate for anamorphic 16:9 (we have sample data on the drive) also we have a software VITC generator thanks to [Unai VITC](https://github.com/unai-d/Unai.VITC) so integrating that at some point in future would be helpful.

DreckSoft, 2023-01-29:
It the WSS used anywhere besides PALplus? 

So this could be a start for https://github.com/happycube/ld-decode/issues/555

harrypm, 2023-01-29:
@DreckSoft Mostly tape media releases, and pretty normal in 2000s era TV recordings, and most HD/HDV era camcorders have WSS on their outputs for composite.