happycube/ld-decode #835: There's a way i can use ld-decode on Windows?

rgm89git, 2023-01-16:
I've been trying to find ways to compile ld-decode on Windows. What options do you guys recommend?

simoninns, 2023-01-16:
I'd recommend installing Linux :)  However, there is someone maintaining a windows build of ld-decode - I would hop onto the discord server or IRC and ask in the general-irc-relay channel. Details for discord/IRC are in the wiki.

closed 2023-01-16 by simoninns

harrypm, 2023-01-16:
@simoninns WSL2 should be noted as it's been working for 2 years just fine, it's all CLI workflow but also has GUI support on windows 11, vhs-decode repo also has a build of the ld-tools suit for windows natively.

But yeah a full Linux box is better for a full-scale capture chain once you work on more than 2 things at the same time lol.