happycube/ld-decode #844: Audio offset for AC3

atsampson, 2023-02-05:
Having watched a few discs with AC3 audio now, it works really well, but lip sync is consistently out with the AC3 track - it needs to be delayed by about 190-200 ms to line up with the EFM and analogue tracks.

The PAL/NTSC LaserVision specs note that EFM decoding will incur a delay, and say that you should advance the EFM audio by 15.3 ms relative to the analogue audio to compensate. So I suspect the LaserDisc AC3 spec (which we don't have) has some similar wording.

Would it be worth having ld-decode take this into account when feeding RF to the AC3 demodulator?

DreckSoft, 2023-02-17:
Absolutely yes. Saves a lot of time in post-processing and what's even more important: It would be less error prune. A lot of people using DDD to preserve movies will miss the necessary delay.

happycube, 2023-05-05:
It's trickier than it sounds since it probably means adding silence to the AC3 file itself.  Preferably silent frames so the main data doesn't need to be re-encoded.

DreckSoft, 2023-05-26:
Yes, if you have an AC3 frame with silence with the same bitrate, you can just copy it at the beginning.

But you can only correct multiples of 32ms, cause that's the duration of one AC3 frame (with 48khz sampling rate).

AkBKukU, 2023-11-09:
I also noticed this, I've been using `ffmpeg` to correct it when making my final re-encoded file. You can add a delay to input sources with `-itsoffset` so I've been using the following with AC3 reads `-itsoffset 0.19 -i *.ac3`.

ValeZAA, 2024-07-13:
There is another issue here. Due to the way lossy codecs work, and AC3 is no exception, there should be a priming of silence in the start, not present in the original source audio on disk creation: 256 samples in the very start and some in the end (end remainder can be absent). Those should be cut. But check in audacity whether AC3 has silence in the start.

Mp4 has edit list for that, ac3 has edts atom with media time 256.

So it is not just 190 ms.

simoninns, 2026-03-31:
AC3 has been moved from the ld-decode repo