happycube/ld-decode #858: Backporch dropout causes TBC failure

Gamnn, 2023-06-12:
I've got a CLV with a scratch, and this happened.

[image]

```
2023-06-12 19:23:23,621 - lddecode - DEBUG - Frame 19/110000: File Frame 19: CLV Timecode 3:56.14 Frame #7094 
2023-06-12 19:23:23,927 - lddecode - DEBUG - Frame 20/110000: File Frame 20: CLV Timecode 3:56.15 Frame #7095 
2023-06-12 19:23:24,038 - lddecode - WARNING - WARNING: TBC failure at line 217
2023-06-12 19:23:24,038 - lddecode - WARNING - WARNING: TBC failure at line 218
2023-06-12 19:23:24,039 - lddecode - WARNING - WARNING: TBC failure at line 219
2023-06-12 19:23:24,040 - lddecode - WARNING - WARNING: TBC failure at line 220
2023-06-12 19:23:24,041 - lddecode - WARNING - WARNING: TBC failure at line 253
2023-06-12 19:23:24,041 - lddecode - WARNING - WARNING: TBC failure at line 254
2023-06-12 19:23:24,041 - lddecode - WARNING - WARNING: TBC failure at line 255
2023-06-12 19:23:24,041 - lddecode - WARNING - WARNING: TBC failure at line 256
2023-06-12 19:23:24,042 - lddecode - WARNING - Analog audio processing error, muting samples
2023-06-12 19:23:24,185 - lddecode - WARNING - At field #40, Field phaseID sequence mismatch (4->3) (player may be paused)
2023-06-12 19:23:24,327 - lddecode - DEBUG - Frame 21/110000: File Frame 21: CLV Timecode 3:56.16 Frame #7096 
2023-06-12 19:23:24,631 - lddecode - DEBUG - Frame 22/110000: File Frame 22: CLV Timecode 3:56.18 Frame #7098 
```

https://mega.nz/file/zYI2AIYI#rzTZ6cq1Xh61n8aG6Hag9CN3bE15FBO8jGTquzeCrTk

happycube, 2023-06-12:
Cut to six frames and added to ld-decode-testdata/issues/issue858.ldf.

(reproduced but not investigated yet)

happycube, 2023-06-18:
As I expected, this is the result of a player skip.  I've improved the error reporting so that one knows it's a skip.

closed 2023-06-20 by happycube