happycube/ld-decode #860: Support MUSE decoding

DreckSoft, 2023-06-18:
I'm just opening this as I think the progress of @staffanu from issue #622 might get lost. 

According to this post: https://forum.videohelp.com/threads/394168-Current-status-of-ld-decode-vhs-decode-(true-backup-of-RF-signals)/page32#post2682590
he already seem to be able to decode a picture.

Unfortunately I do not have a single MUSE disc, so I can't test or provide a sample.

staffanu, 2023-06-18:
Hi,

Just for information I've been working a bit more on muse decoding. There are improvements to be done in filtering, and especially motion detection is very early code, mostly a placeholder really, but the output from motion detection is handled reasonably.

I spent the last couple of months rewriting what I have in Scala into c++ and gpu (using Vulkan) and can now decode in real time on my laptop. Working on audio right now:)

I'll post something on Discord when there is something to test or help out with: the current code is too much of a moving target, but it is in my bitbucket repo (same as the ac3 stuff) if someone wants to have a look.

Staffan

DreckSoft, 2023-06-20:
Wow, so the muse decoding is (much) faster than PAL / NTSC. 

I'll definitely follow your progress. Still trying to get a MUSE disc that doesn't cost a fortune. 

How did you capture the disc? With a regular DDD?

staffanu, 2023-06-20:
Yes, running it on the GPU does help a lot -- the scala code takes forever (like overnight for a few minutes), but it is not written for speed at all, so the comparison is not a fair comparison between Scala and GLSL. I made a reasonably optimized c++ version in between, but that was too slow for real-time decoding, if only a factor of 5 or so (single threaded, so could be possible with many cores). Since I'm aiming for cheap hardware I want the GPU route.

I captured the MUSE output from the player using a custom FPGA controlled sampling circuit similar to the DDD, but that adjusts the sampling phase to the horizontal sync of the MUSE signal (the way described in the specification). But, as long as all the signal is captured (i.e., everything over 8.1 MHz), it should be possible to reconstruct the signal at the correct instances, so I'm going to try to write something that reads DDD recordings.

If we want to sample the RF signal using the DDD I think the low-pass filter might have to be altered a bit, but the sampling frequency might be enough, but just barely.

Zipdox2, 2025-01-28:
I have no LD or MUSE anything but I find this topic interesting. I don't know if this is helpful at all but these are the standards documents that define MUSE.
[Rec. ITU-R BO.786.pdf](https://github.com/user-attachments/files/18567515/Rec.ITU-R.BO.786.pdf)
[SMPTE 240M-1995.pdf](https://github.com/user-attachments/files/18567517/SMPTE.240M-1995.pdf)
[SMPTE 260M-1999.pdf](https://github.com/user-attachments/files/18567516/SMPTE.260M-1999.pdf)