happycube/ld-decode #866: --threads option not working

ArcadeHacker, 2023-09-17:
Hi there, i'm unable to get --threads to work as intended on a high count core machine, no matter what I do I cannot get ld-decode to run more than aprox. 6 threads. Tested on a computer with 36 threads, other tools such as sysbench show proper behavior on the same computer (eg use all available threads). Tested on Ubuntu 22.04.3 (intel). 

Thanks.

happycube, 2023-09-17:
The TBC is a bottleneck at that point, so the worker threads that do the RF decoding have nothing to do.

harrypm, 2023-09-18:
@ArcadeHacker Its all single core bais speed wise, so no more then 4-6 threads will improve anything across the formats though [speed testing is tracked here](https://github.com/oyvindln/vhs-decode/wiki/Speed-Testing) curreent gen hardware can decode some formats upto 16fps+

happycube, 2023-10-22:
It'd be nice to improve this (it's better than it was) but improving this is a side effect of TBC optimization/threading, closing.

closed 2023-10-22 by happycube