happycube/ld-decode #918: efm-decoder-f2: SectionTime invalid frame value crash

JoeyLemur, 2025-03-18:
### Description

This disc (disc 1 of https://www.lddb.com/laserdisc/24880/G128F0083/Last-Starfighter-The) fails with "Fatal: SectionTime::SectionTime(): Invalid frame value of -269999" when run against efm-decoder-f2

### Steps to Reproduce

1. `ld-decode --start 350 --length 1800 --threads 6 --ntsc the_last_starfighter_s1.flac.ldf test/output`
2. `cd test`
3. `efm-decoder-f2 --debug --show-all-debug output.efm output.f2`

### Actual Behaviour

Gets as far as "F2 Section Metadata Correction statistics" before crashing:
```
Info: F2 Section Metadata Correction statistics:
Info:   F2 Sections:
Info:     Total: 0 (0 F2)
Info:     Corrected: 0
Info:     Uncorrectable: 0
Info:     Pre-Leadin: 4504
Info:     Missing: 0
Info:     Padding: 0
Info:     Out of order: 0
Info:   QMode Sections:
Info:     QMode 1 (CD Data): 0
Info:     QMode 2 (Catalogue No.): 0
Info:     QMode 3 (ISO 3901 ISRC): 0
Info:     QMode 4 (LD Data): 0
Info:   Absolute Time:
Info:     Start time: 59:59:74
Info:     End time: 00:00:00
Fatal: SectionTime::SectionTime(): Invalid frame value of -269999
```

No output is written to `output.f2`

### Environment
- `efm-decoder-f2 Branch: main / Commit: 5d339d9d`
- MacOS 15.3.2 (MacBook Air M2, 2022)

JoeyLemur, 2025-03-18:
[output.efm.gz](https://github.com/user-attachments/files/19321714/output.efm.gz)

[debug.log](https://github.com/user-attachments/files/19321716/debug.log)

JoeyLemur, 2025-03-18:
Neglected to mention the actual crash error:  `Abort trap: 6`

JoeyLemur, 2025-03-18:
Different disc, same issue.

[test2_debug.log](https://github.com/user-attachments/files/19326258/test2_debug.log)
[test2.efm.gz](https://github.com/user-attachments/files/19326257/test2.efm.gz)

boysrbackintown, 2025-06-23:
Having similar issues with multiple discs, usually early ones that I would otherwise have to use the -t flag with. Like you I got one that stated "Fatal: SectionTime::SectionTime(): Invalid frame value of -269999" at the end.

```
Info: Progress: 100 %
Info: Flushing decoding pipelines
Info: Processing final pipeline data
Info: Decoding complete
Info: T-values to Channel Frame statistics:
Info:   T-Values:
Info:     Consumed: 2317302012
Info:     Discarded: 3541
Info:   Channel frames:
Info:     Total: 18985512
Info:     588 bits: 18980658
Info:     >588 bits: 2417
Info:     <588 bits: 2437
Info:   Sync headers:
Info:     Good syncs: 18960024
Info:     Overshoots: 12630
Info:     Undershoots: 256
Info:     Guessed: 12602
Info:
Info: Channel to F3 Frame statistics:
Info:   Channel Frames:
Info:     Total: 18985512
Info:     Good: 18980685
Info:     Undershoot: 2419
Info:     Overshoot: 2408
Info:   EFM symbols:
Info:     Valid: 607528167
Info:     Invalid: 7989
Info:   Subcode symbols:
Info:     Valid: 18985462
Info:     Invalid: 50
Info:
Info: F3 Frame to F2 Section statistics:
Info:   F3 Frames:
Info:     Input frames: 18985512
Info:     Good sync0 frames: 193591
Info:     Missing sync0 frames: 91
Info:     Undershoot sync0 frames: 130
Info:     Overshoot sync0 frames: 10
Info:     Lost sync: 0
Info:   Frame loss:
Info:     Presync discarded F3 frames: 0
Info:     Discarded F3 frames: 885
Info:     Padded F3 frames: 123
Info:
Info: F2 Section Metadata Correction statistics:
Info:   F2 Sections:
Info:     Total: 0 (0 F2)
Info:     Corrected: 0
Info:     Uncorrectable: 0
Info:     Pre-Leadin: 193719
Info:     Missing: 0
Info:     Padding: 0
Info:     Out of order: 0
Info:   QMode Sections:
Info:     QMode 1 (CD Data): 0
Info:     QMode 2 (Catalogue No.): 0
Info:     QMode 3 (ISO 3901 ISRC): 0
Info:     QMode 4 (LD Data): 0
Info:   Absolute Time:
Info:     Start time: 59:59:74
Info:     End time: 00:00:00
Fatal: SectionTime::SectionTime(): Invalid frame value of -269999
Aborted (core dumped)
```

closed 2025-11-08 by simoninns