happycube/ld-decode #925: TAK Compression Support

smiRaphi, 2025-05-31:
### Is your proposal related to a problem?

Rips taking up a lot of space.

### Is your proposal already in the issue tickets?

No

### Is your proposal related to tbc-video-export?

No

### Describe the solution you'd like

Using a better codec (eg [TAK](http://thbeck.de/Tak/Tak.html)).

### Additional context

I've did some testing and TAK is about 25-30% better compression wise and also faster to encode while it's a bit slower on the decode side but I would say that you don't use these files too often and that it's more for archiving so compression is worth more than speed as long as it doesn't take ages imo.

### Contribution

I am gonna look into it but I don't think I know enough C++, the project & about the TAK SDK to do it myself.

smiRaphi, 2025-05-31:
Ah right, sorry, haven't thought of the closed source thing, I always forget that with TAK. But TAK is known to be better than FLAC and I tested it using some rips available online, for example https://archive.org/details/dialog-2001-vhs-capture and then encoded parts of 25 hours (because of the 4GB thing but I think it would be pretty easy to make a custom format that just stores multiple parts 4GB tak data and you can also just split & stitch the file for en/decoding) in both flac & tak and compared the 2.

smiRaphi, 2025-05-31:
But even just the ability to just read TAK files (or an aforementioned custom format using TAK) would be amazing to have because the user can just compress the WAV files themselves with the encoder that's available to download

oyvindln, 2025-06-01:
You would have to make some external tool to pipe the output from ddd to if you want support for this codec since it is limited to windows and the license is incompatible. It doesn't seem very practical. 

It seems ffmpeg [supports](https://ffmpeg.org/doxygen/trunk/libavcodec_2takdec_8c_source.html) decoding the format but I have no idea if it's compatible with the latest version of the format or not.

If we want a replacement for FLAC it really needs to be something that is free software.

smiRaphi, 2025-06-01:
Like I said, not talking about a replacement as long as it's closed source, just as a better alternative compression wise and I am making such a tool right now.
Also there will need to be custom support for it and not just ffmpeg because TAK has a 4GB limit.

simoninns, 2025-12-02:
Closing issue as out of scope - the project uses open source FLAC.

closed 2025-12-02 by simoninns